pop-os/firmware-manager #1: Design

WatchMkr, 2019-06-19:
Integrate automatic firmware updates into Pop!_OS via a system library that displays in gcc (Devices > Firmware) and daemon that checks for updates and provides notifications. The library and daemon can be installed on other distributions for system76 firmware updates. Both will be installed for system76 customers using Ubuntu probably with the system76-driver package.

[image]

"System Firmware" is firmware that requires a reboot to install.

[image]

[image]

mmstick, 2019-06-24:
Tracking progress:

- [x] system76-firmware client in Rust
  - [ ] changes are merged into the system76-firmware repo
- [ ] GTK library
  - [x] Rust library is written
  - [ ] C bindings to the library is written
  - [ ] Firmware update dialog is written
  - [ ] Integration in GNOME Software
- [x] GTK application in Rust based on the library

mmstick, 2019-06-24:
The new firmware widget library is working in Rust now. Any errors reported by the firmware client accessing the firmware daemon are also being handled and directed to an info bar in the widget.

[image]

[image]

WatchMkr, 2019-06-24:
Looks good! Check with @isantop on the dialog box icon. We have a Pop firmware icon for that image.

mmstick, 2019-06-25:
The C bindings are complete, with a [working example](https://github.com/pop-os/firmware-manager/tree/master/ffi/examples/c), so now I'll begin working on integration into GNOME Settings.

mmstick, 2019-06-25:
:rocket: The firmware panel in GNOME Settings worked on the first try.

[image]

mmstick, 2019-06-25:
Screenshot when the firmware daemon service is not active, and thus firmware scanning is deactive.

[image]

closed 2019-08-16 by WatchMkr