pop-os/firmware-manager #2: Transition from system76-driver's firmware UI

jackpot51, 2019-07-18:
- [ ] Release fwupd with privacy concerns addressed (@jackpot51)

- [ ] Release firmware-manager, but do not depend on it yet (@mmstick)
  - [ ] Add autostart file to show notifications by running firmware-manager in the background without a UI
  - [ ] The notification should launch the gnome-control-center firmware pane if the desktop file for it can be found, and launch the firmware-manager application otherwise

- [ ] Perform testing on firmware-manager (@brs17)
  - [ ] Find a way to roll back or simulate fwupd updates
  - [ ] Test update notifications
  - [ ] Test use in Ubuntu or alternative Desktop Environments

- [ ] Prepare a branch of gnome-control-center with the following (@mmstick):
  - [ ] Firmware pane that shows up in GNOME Shell
  - [ ] Hide firmware-manager from GNOME Shell, potentially by using dpkg-divert or NoDisplay=True if and only if gnome-control-center with a Firmware pane is installed

- [ ] Prepare a branch of system76-driver with the following (@jackpot51):
  - [ ] Depends on firmware-manager
  - [ ] Remove system76-firmware tool
  - [ ] Remove desktop file and autostart

mmstick, 2019-07-18:
Currently, the firmware-manager application & desktop file is in its own package, separate from the GTK lib package which GNOME Control Center uses. So, Pop!_OS could depend on the lib with the patched GNOME Settings, and then Ubuntu would install the `firmware-manager` package which provides the standalone application.

mmstick, 2019-07-18:
> Find a way to roll back or simulate fwupd updates

I added a [TESTING.md](https://github.com/pop-os/firmware-manager/blob/master/TESTING.md#downgrading-fwupd-managed-firmware) which shows how to do this.

mmstick, 2019-07-18:
Firmware manager is now ready to release, as soon as the updated `fwupd` package is released, and the debian packages for this are built.

- The `firmware-manager-notify` package contains the autostart file, and notification binary. This can be added as a depends for Ubuntu and Pop!_O. It will attempt to exec `gnome-control-center firmware` if `/usr/share/applications/gnome-firmware-panel.desktop` is found, else the standalone application.
- The `firmware-manager` package contains the standalone GTK application and its desktop file.
- The `libfirmware-manager` package provides the GTK widget library used as a depends by GNOME Settings

closed 2019-08-16 by mmstick