pop-os/firmware-manager #13: Discussion point: How frequent should the user be notified of available firmware?

brs17, 2019-07-29:


WatchMkr, 2019-07-29:
Firmware often includes security fixes. We should check for updates daily and notify whenever they're available.

Because th firmware device list is small and updates infrequent, it is unlikely we'll frustrate users with too many notifications.

mmstick, 2019-07-29:
We would need to determine which application starts the notify application, since it only runs long enough to check for updates before exiting. Or if it should be its own user service.

WatchMkr, 2019-07-29:
Seems that `firmware-manager-notify` is a natural fit. What do you think though?

mmstick, 2019-07-29:
`firmware-manager-notify` is executed by the login service at startup, but it exits as soon as soon as it is clicked. We would need to either have an existing background service occasionally launch the notify application to check for updates, or we would keep it running in the background permanently, to occasionally check for updates itself.

mmstick, 2019-07-30:
I think I'll do this by adding a systemd user timer to the notify package, and have that set to rerun the notification binary every day.

mmstick, 2019-07-31:
This should be working now.

closed 2019-08-05 by WatchMkr