pop-os/firmware-manager #18: Firmware changelog needs to be better formatted

brs17, 2019-07-31:
Kate will provide guidance in how to best format the firmware changelog as seen below:
[image]

WatchMkr, 2019-07-31:
Reviewed w/ Kate: Hierarchy Update

[image]

brs17, 2019-08-01:
The dropdown notification looks good.

This is what the layout and changelog currently looks like.
[image]

 I think there's still a few changes that we want to include before we're ready to ship. 
 - The mockup includes a <b>Changelog</b> title as the first text in the expanded dropdown.
 - Separators should be between changelog entries not after the version provided in a specific entry.

mmstick, 2019-08-01:
Updated look

[image]

commit 85240d47 (2019-07-31): :sparkles: Set margins for changelogs in dropdown

Closes #18
closed 2019-08-02 by brs17