pop-os/firmware-manager #19: Multiple thelio-io boards only show one `Update` button

brs17, 2019-07-31:
On systems with multiple thelio-io boards, there is only one `Update` button provided. If a separate update button cannot be provided for each board, potentially both boards should be listed as one device.

[image]

mmstick, 2019-07-31:
The reason for multiple to display is because they may have different versions of firmware installed, theoretically.

WatchMkr, 2019-07-31:
We discussed and decided to collapse multiple boards to one entry and show the oldest firmware version. Updating always installs on all Thelio Io boards.

commit 0a32202e (2019-07-31): :sparkles: Collapse Thelio I/O devices into one

Closes #19
commit 4cda3a69 (2019-07-31): :sparkles: Collapse Thelio I/O devices into one

Closes #19
closed 2019-07-31 by mmstick