pop-os/firmware-manager #21: System firmware does not need a progress bar

brs17, 2019-07-31:
With system firmware already downloaded before notifying the user of new firmware, the progress bar shown after selecting reboot shows up for a split second providing no real value to the user.

This progress bar c/should probably just be removed from system firmware updates.

mmstick, 2019-07-31:
Should we still display a progress bar when a device is being flashed, and system firmware is then clicked to schedule an update in the queue? Currently, devices are being flashed serially, so firmware will wait in line for their turn to be processed.

brs17, 2019-07-31:
Per discussion in person, we decided we will keep the progress bar but change the wording to "Waiting". With that change and fixing https://github.com/pop-os/firmware-manager/issues/22 it will look good. 

The reason for the progress bar is for queuing multiple firmware updates (and preventing a reboot in the middle of a device firmware update).

mmstick, 2019-07-31:
That reminds me that we should activate a lock/suspend/reboot/shutdown inhibitor

brs17, 2019-08-02:
I think we're 90% there. I'm thinking we don't need the progress bar underneath `Waiting` as that progress bar never really progresses as once waiting is over the system reboots.

[image]

commit bdb58c7b (2019-07-31): :sparkles: Default progress bars to "Waiting" after click

Closes #21
commit 700cb890 (2019-08-02): :sparkles: Add a waiting label widget to device widget stack

Fixes #21
commit f955d603 (2019-08-02): :sparkles: Add a waiting label widget to device widget stack

Fixes #21
closed 2019-08-02 by mmstick