pop-os/firmware-manager #23: Connect to A/C message should be smart about when to be displayed

brs17, 2019-07-31:
Currently on all system firmware updates, there is a message at the bottom of the pop-up menu that tells the user if the machine they are on is a laptop, to plug in the laptop before running the update. 

It should be possible to determine if the machine is a laptop or desktop and only show a message when the machine is a laptop.

brs17, 2019-08-01:
1. I think the message text could be shrunk from `"If you're on a laptop, plug into power before you begin."` to `Plug into power before you begin` since the message will only be shown on laptops.
2. I am still getting the message on a thelio-major-r1. The output of `upower -d` is: 
```
Device: /org/freedesktop/UPower/devices/DisplayDevice
  power supply:         no
  updated:              Thu 01 Aug 2019 03:30:28 PM MDT (184 seconds ago)
  has history:          no
  has statistics:       no
  unknown
    warning-level:       none
    icon-name:          'battery-missing-symbolic'

Daemon:
  daemon-version:  0.99.10
  on-battery:      no
  lid-is-closed:   no
  lid-is-present:  no
  critical-action: HybridSleep
```

mmstick, 2019-08-01:
The latest commit should fix this

brs17, 2019-08-02:
With this logic implemented, the notification only appears when on a laptop that is running on battery. I am wondering if a message should show when on a laptop that is plugged in should say something along the lines of "Keep your laptop plugged in during this update". Thoughts @WatchMkr ?

WatchMkr, 2019-08-02:
Let's stick with the plug in message if you're not plugged in.

commit 3ff0502a (2019-07-31): :sparkles: Only show battery message when a battery is detected

Closes #23
closed 2019-08-02 by WatchMkr