pop-os/firmware-manager #26: Ensure operability between firmware-manager and GNOME Software

brs17, 2019-08-01:
This has not been tested yet, but we'll want to ensure good behavior between fwupd firmware updates/notifications from the firmware-manager and GNOME Software.

WatchMkr, 2019-08-05:
Add build flag to disable firmware-manager fwupd updates for Ubuntu.

mmstick, 2019-08-05:
Luckily I foresaw this in advance, and both `system76-firmware` and `fwupd` are conditional compilation features. However, I just noticed that the flags aren't working right. Will fix that now.

mmstick, 2019-08-05:
An Ubuntu-specific branch can now use `make features=system76` to build everything with system76-support only, whereas packaging for Pop!_OS can use `make features='fwupd system76'` to build for both.

jackpot51, 2019-08-05:
@mmstick we could set some environmental variable when doing CI builds of packages for Ubuntu - i.e. when --dev is passed to the ci script.

jackpot51, 2019-08-05:
Then use that variable to determine the default features

mmstick, 2019-08-05:
I've added a `S76_DEV` variable to the makefile. If it is set to `1` it will default to `system76`, otherwise `fwupd system76`.

WatchMkr, 2019-08-16:
At the moment, fwupd devices are visible in firmware-manager in Ubuntu 18.04. However, this is okay because it appears fwupd updates are not showing up in Ubuntu Software on 18.04.

closed 2019-08-16 by WatchMkr