pop-os/firmware-manager #28: Firmware manager page is blank when the system is offline

brs17, 2019-08-02:
When there is no network connection, currently an error appears `error trying to connect`.

When a system is offline and not connected to a network, this page should still be able to show the currently installed firmware (just not be able to check for updates).

mmstick, 2019-08-02:
I think we'll need to update `system76-firmware` to return dbus error types instead of strings, as there's currently no way to determine the kind of error received.

brs17, 2019-08-02:
I'm wondering, what errors could we get from system76-firmware that we would want to show to the user? Potentially we just handle errors where we want to change the displayed output and ignore other ones?

Example thoughts:
Error 1: No network connection
Workaround A: Ignore error, show currently installed firmware
Workaround B: Show currently installed firmware, provide some sort of message that a network connection is necessary for searching for firmware updates

Error 2: error calling the download method, which provides the changelogs
Workaround A: When the error is received, display `Changelog missing` or something to that extent. Would part of the `Changelog missing` message want to include something about how there is an error?
Workaround B: Leave changelog blank.

Other than not having a network connection (and errors gathered from that), I am not sure what value an error could provide a user.

Thoughts on how to deal with errors @WatchMkr / @mmstick / @jackpot51 ?

commit 2e5e903c (2019-08-05): :sparkles: Handle network connecction errors

If a network connection cannot be stablished to fetch firmware changelogs, we will instead display the devices with or without those changelogs

Fixes #28
closed 2019-08-05 by mmstick