pop-os/firmware-manager #36: On unsupported hardware, application remains blank

brs17, 2019-08-16:
ISSUE: 
Looking at the firmware tab in GNOME Control Center on 19.04 on a machine without any supported System76 or fwupd supported devices the firmware panel remains blank.

[image]

HYPOTHESIS: It appears that the search for fwupd devices begins, as evident by the `scanning fwupd devices` message. The search never appears to end (perhaps because `TIMEOUT=-1` in fwupd-dbus/src/lib.rs ?) and therefore the `view_empty` state is never reached and the panel remains blank.

mmstick, 2019-08-16:
That could be what's happening, but if so, that would be an issue with the fwupd daemon holding the connection because it got stuck, and therefore isn't closing the socket, and then won't be able to respond to any future requests.

I will check if it could be an issue with the "No devices" view simply never getting shown. I will also add a debug message so that you can see if the fwupd connection has cleanly exited.

mmstick, 2019-08-16:
It would seem that I never wrote the code that sends `FirmwareSignal::ScanningComplete` back to the main thread, so the UI continued to assume that updates were still being scanned.

brs17, 2019-08-16:
This appears to still be an issue even after applying the appropriate patch.

mmstick, 2019-08-16:
I got it working again.

commit 914900d3 (2019-08-16): fix(core): Tell the main thread when scanning is complete

Seems that I had forgotten to send this signal when scanning is complete, so the UI assumed that scanning never completed.

Closes #36
commit 57879f7c (2019-08-16): fix(core): Tell the main thread when scanning is complete

Seems that I had forgotten to send this signal when scanning is complete, so the UI assumed that scanning never completed.

Closes #36
commit 703061d2 (2019-08-16): fix(gtk): Ensure that the empty view is shown when set

Really closes #36
commit c0e1f1f3 (2019-08-16): fix(gtk): Ensure that the empty view is shown when set

Really closes #36
closed 2019-08-16 by mmstick