pop-os/firmware-manager #38: Phrasing of system upgrade pop-up is strange

brs17, 2019-08-16:
Currently the pop-up for system firmware notifications says: `Fixes and features include:`  and then lists the changelog. Potentially that is a little odd, as showing the changelog of versions prior to what is installed isn't really showing what fixes and features are included in the update. 

[image]

Maybe change the wording to just state that the following is a changelog? Or show only the changelog containing the latest update?

WatchMkr, 2019-08-16:
I agree. Let’s remove that line entirely and match the new change log design on the main panel.

mmstick, 2019-08-16:
Like so, or should we have the padding for dialogs?

[image]

brs17, 2019-08-16:
Keep "Firmware version $version is available." above *Changelog*.

commit e2822f97 (2019-08-16): feat(gtk): Dialogs now match the changelog revealer layout

Closes #38
commit da119bac (2019-08-16): feat(gtk): Dialogs now match the changelog revealer layout

Closes #38
closed 2019-08-16 by mmstick