pop-os/firmware-manager #42: Filtering supported System76 models

brs17, 2019-08-16:
Currently in the system76-driver, we filter out products that do not support updates via this process.(https://github.com/pop-os/system76-driver/blob/master/system76driver/firmware.py#L43). We do not have this for firmware-manager. This therefore includes all System76 products even if they should not be shown (as updates would never arrive through this mechanism).

mmstick, 2019-08-16:
Would this be needed in the firmware-manager, or should that go in the system76-firmware daemon?

jackpot51, 2019-08-16:
It would probably be better to go in system76-firmware

mmstick, 2019-08-16:
I'll add it then

jackpot51, 2019-08-16:
Thanks! It should return errors from basically every DBUS function if it isn't in the whitelist

mmstick, 2019-08-16:
The `firmware-client` branch was just updated: https://github.com/pop-os/system76-firmware/pull/15/commits/b9c3f8616b837ba52ff85ec95b46e5d5586d1584

WatchMkr, 2019-08-16:
Looks good

closed 2019-08-16 by WatchMkr