pop-os/firmware-manager #44: Pop!_OS 18.04: Firmware icon missing on System firmware pop-up

brs17, 2019-08-16:
On Pop!_OS 18.04, the system firmware pop-up is missing the appropriate firmware icon

[image]

mmstick, 2019-08-16:
That's strange. The [firmware icons were added to the bionic branch of our icon theme](https://github.com/pop-os/icon-theme/commits/master_bionic).

isantop, 2019-08-16:
@mmstick @brs17 Unless we're specifically overriding the Icon theme, that is correct. `application-x-firmware` in the bionic icon theme is symlinked to the generic binary file icon (which that is).

isantop, 2019-08-16:
It's my vote that we not override the icon theme, and thus this is expected behavior.

isantop, 2019-08-16:
It's pulling the correct icon. It's being overridden by the Pop overrides folder, which is where that change should have been made in the first place. 

The correction is in place as pop-os/icon-theme#48

brs17, 2019-08-16:
Fixed, thank you @isantop !

closed 2019-08-16 by brs17