pop-os/firmware-manager #47: Minimum version of Rust required to build?

rklrkl, 2019-08-18:
Looking at README.md, I see no mention of what the minimum version of Rust is required to build Firmware Manager.

I just tried building it on CentOS 7 (after a "yum install cargo") and cargo threw out an error:

cargo run -p tools --bin desktop-entry  -- \
	--appid com.system76.FirmwareManager \
	--name "Firmware Manager" \
	--icon firmware-manager \
	--comment "Manage system and device firmware" \
	--keywords firmware \
	--keywords system76 \
	--keywords fwupd \
	--keywords lvfs \
	--categories System \
	--categories GTK \
	--binary com.system76.FirmwareManager \
	--prefix /usr \
	--startup-notify
error: failed to parse manifest at `/tmp/firmware-manager-master/gtk/Cargo.toml`

Caused by:
  could not parse input as TOML

Caused by:
  unexpected character found: `\\` at line 22

EPEL for CentOS 7 only has Rust 1.36.0, which presumably isn't recent enough - could you please add the minimum Rust version requirement to README.md?

rklrkl, 2019-08-18:
OK, I tracked this one down shortly after posting up and it's not the Rust version that's the problem - it's the Makefile's use of echo that's causing the build failure.

At lines 80, 88 and 96, the Makefile has this code:

echo -n '\nfirmware-manager ...

This is incorrect quoting - single quotes override any backslash escaping and put the literal backslash and n in the output instead. The fix I tried (that worked) was to remove the \n and put it as a separate echo statement of its own on the line above. So for example, line 80/81 becomes:

 echo >> gtk/Cargo.toml
 echo -n 'firmware-manager...

Once I fixed the 3 sections of the Makefile with the problem, I got further (but hit a git snag almost straight away that I'll post up as a separate issue).

mmstick, 2019-08-18:
Rust projects define the minimum Rust version through the [rust-toolchain file](https://github.com/pop-os/firmware-manager/blob/master/rust-toolchain)

closed 2019-08-18 by mmstick

yochananmarqos, 2019-08-20:
Why is this issue closed when it was not addressed?
```
echo -n '\nfirmware-manager = { path = "../", default-features = false, features = [ ' >> gtk/Cargo.toml
echo -n '\nfirmware-manager-gtk = { path = "../", default-features = false, features = [ ' >> gtk/ffi/Cargo.toml
echo -n '\nfirmware-manager = { path = "../", default-features = false, features = [ ' >> notify/Cargo.toml
cargo run -p tools --bin desktop-entry  -- \
	--appid com.system76.FirmwareManager.Notify \
	--name "Firmware Manager Check" \
	--icon firmware-manager \
	--comment "Check for firmware updates, and display notification if found" \
	--categories System \
	--binary com.system76.FirmwareManager.Notify \
	--prefix /usr \

for feature in fwupd; do \
	echo -n "\"$feature\"," >> gtk/Cargo.toml; \
done
for feature in fwupd; do \
	echo -n "\"$feature\"," >> gtk/ffi/Cargo.toml; \
done
cargo build --manifest-path gtk/ffi/Cargo.toml --no-default-features --release
for feature in fwupd; do \
	echo -n "\"$feature\"," >> notify/Cargo.toml; \
done
cargo run -p tools --bin pkgconfig  -- \
	firmware_manager /usr/lib /usr/include
echo ' ] }' >> gtk/Cargo.toml
echo ' ] }' >> gtk/ffi/Cargo.toml
echo ' ] }' >> notify/Cargo.toml
error: failed to parse manifest at `/home/yochanan/Documents/pkgbuilds/firmware-manager-git/src/firmware-manager/gtk/Cargo.toml`

Caused by:
  could not parse input as TOML

Caused by:
  unexpected character found: `\\` at line 27
error: failed to parse manifest at `/home/yochanan/Documents/pkgbuilds/firmware-manager-git/src/firmware-manager/gtk/Cargo.toml`

Caused by:
  could not parse input as TOML

Caused by:
  unexpected character found: `\\` at line 27
error: failed to parse manifest at `/home/yochanan/Documents/pkgbuilds/firmware-manager-git/src/firmware-manager/gtk/ffi/Cargo.toml`

Caused by:
  could not parse input as TOML

Caused by:
  unexpected character found: `\\` at line 15
make: *** [Makefile:113: target/com.system76.FirmwareManager.desktop] Error 101
make: *** Waiting for unfinished jobs....
make: *** [Makefile:129: target/com.system76.FirmwareManager.Notify.desktop] Error 101
make: *** [Makefile:143: target/release/libfirmware_manager.so] Error 101
error: failed to parse manifest at `/home/yochanan/Documents/pkgbuilds/firmware-manager-git/src/firmware-manager/gtk/Cargo.toml`

Caused by:
  could not parse input as TOML

Caused by:
  unexpected character found: `\\` at line 27
make: *** [Makefile:148: target/firmware_manager.pc] Error 101
```

Line 27 ends up like this:
```
yansi = "0.5.0"\nfirmware-manager = { path = "../", default-features = false, features = [ "fwupd", ] }
```

mmstick, 2019-08-20:
What shell does your Make use by default? It works on our build server, and on my two development machines. Otherwise we wouldn't have packages being built.

yochananmarqos, 2019-08-20:
I'm using `make` 4.2.1-3 on Manjaro GNOME. The default shell is `bash` 5.0.007-1.

mmstick, 2019-08-20:
See if the latest commit fixes your issue. The default shell used on the debian platform is `sh` / `dash` for makefiles.

yochananmarqos, 2019-08-20:
Yes, that worked. Thank you.