pop-os/firmware-manager #48: Build from source not finding git repo?

rklrkl, 2019-08-18:
I'm on CentOS 7 (Rust 1.36.0 which I think is recent enough) and after a Makefile tweak I mentioned in another issue (#47), I'm hitting a git access issue that stops the build immediately. I ran both:

make prefix=/usr features='fwupd system76'

and

make prefix=/usr features='fwupd'

(I don't have a System76 system, hence trying the second version as well) and both resulted in git access errors:

    Updating git repository `https://github.com/pop-os/system76-firmware`
error: failed to load source for a dependency on `system76-firmware-daemon`

Caused by:
  Unable to update https://github.com/pop-os/system76-firmware?branch=firmware-client#f77915c0

Caused by:
  revspec 'f77915c0a3d23b7f4db0473e5956173fbecdc4eb' not found; class=Reference (4); code=NotFound (-3)

This is after checking out the source both with "git clone" and also just downloading the master zip file from Github and unpacking that. It appears that whatever git repo the build is expecting isn't there.

BTW, with the features='fwupd' build, why is it still trying to access the System76 code?

mmstick, 2019-08-18:
Ah, that repository branch no longer exists. It needs to point to the master branch of system76-driver.

mmstick, 2019-08-18:
> BTW, with the features='fwupd' build, why is it still trying to access the System76 code?

Vendoring still requires that the sources are fetched in case you do build the code later.

commit 9e975855 (2019-08-18): chore(deps): Update dependencies

Closes #48
commit b88e98bf (2019-08-18): chore(deps): Update dependencies

Closes #48
commit 97adcb18 (2019-08-19): chore(deps): Update dependencies

Closes #48
closed 2019-08-18 by mmstick