pop-os/firmware-manager #53: Managed Firmware Unavailable

yochananmarqos, 2019-08-20:
Hardware that `fwupd` detects is not detectable by `firmware-manager`. I build it with the `features='fwupd'` make flag on Manjaro GNOME.

`firmware-manager` displays:
> Managed Firmware Unavailable
> No devices supporting automatic firmware updates detected

```
$ fwupdmgr get-devices
Inspiron 5575 System Firmware
  DeviceId:             abded4871dab480db889303c980b43ffb7ca55e2
  Guid:                 165c3bfc-46b9-36e5-2a89-5cae27fed31b
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.1.3.2
  VersionLowest:        0.0.4.4
  VersionFormat:        quad
  Icon:                 computer
  Created:              2019-08-20

Unifying Receiver
  DeviceId:             23f03b8c7695b6d35877233fe7f4ba355b1e78b6
  Guid:                 cc4cbfa9-bf9d-540b-b92b-172ce31013c1
  Guid:                 279ed287-3607-549e-bacc-f873bb9838c4
  Summary:              A miniaturised USB wireless receiver
  Plugin:               unifying
  Flags:                updatable|supported|registered
  Vendor:               Logitech
  VendorId:             USB:0x046D
  Version:              RQR24.07_B0030
  VersionBootloader:    BOT03.03_B0009
  VersionFormat:        plain
  Icon:                 preferences-desktop-keyboard
  InstallDuration:      7
  Created:              2019-08-20

WDC WD10SPZX-75Z10T2
  DeviceId:             f83f14a5b4a3ab547fd6163714b21853eeeda2ba
  Guid:                 febf540f-1153-54e3-ac56-ca0d9af64b30
  Guid:                 ff3edd69-3118-58ea-b493-48ae036d2185
  Guid:                 014f4a74-fe12-56de-8f58-e61b45232cb2
  Summary:              ATA Drive
  Plugin:               ata
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              03.01A03
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-08-20

Samsung SSD 970 EVO Plus 250GB
  DeviceId:             71b677ca0f1bc2c5b804fa1d59e52064ce589293
  Guid:                 47335265-a509-51f7-841e-1c94911af66b
  Guid:                 0a385028-4a26-5bc4-8282-40889897c5e8
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Vendor:               Samsung Electronics Co Ltd
  VendorId:             NVME:0x144D
  Version:              1B2QEXM7
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2019-08-20
```

mmstick, 2019-08-20:
Which version of `fwupd` do you have (we are using `1.2.9`). This would mean that the fwupd daemon did not return any releases for devices with the `supported` flag.

mmstick, 2019-08-20:
The latest commit should display the error that you are receiving, if there is an error.

yochananmarqos, 2019-08-20:
I'm using `fwupd` 1.2.10 and 1.2.9 also worked fine.

Yes, the latest commit shows the same error message that `fwupdmgr get-releases` / `get-updates` does:
```
[ERROR] firmware_manager src/fwupd.rs:54: failure to get fwupd releases for Unifying Receiver: calling GetReleases method failed: "No releases found for device: Not compatible with bootloader version: failed predicate [BOT03.0[0-1]_* regex BOT03.03_B0009]"
```

It's a known issue and there's an [open issue](https://github.com/hughsie/fwupd/issues/1244) about it.

mmstick, 2019-08-20:
We only show devices in the UI when there are firmware releases available for it, so this seems to be working as intended. Still, it's good to be showing the error message from fwupd now.

yochananmarqos, 2019-08-20:
Oh, I see. I was hoping it would show compatible devices as well as if there was an update or not.

mmstick, 2019-08-20:
Technically they're not being managed by fwupd, so they're only theoretically compatible. fwupd will list devices even if there are no firmware releases available for them.

Each `Release` for a device defines a specific firmware release, including the filename of the firmware, changelog details, and other assorted information. Without any releases, there is no way to determine if there are firmware updates, because there is no firmware release to compare to. We know when you have updates when there are releases newer than the version of firmware that you have installed. We still generate changelogs for a device that display the details of each version prior to the one that you have installed, though.

yochananmarqos, 2019-08-20:
That makes sense, thank you for the response.

closed 2019-08-20 by yochananmarqos