pop-os/firmware-manager #58: How to install properly on ubuntu?

jpritikin, 2019-08-29:
I have,

```
deb http://ppa.launchpad.net/system76-dev/stable/ubuntu disco main
```

To get the latest firmware and updates, I need to do what? `apt install firmware-manager`? Do I need to replace `fwupd-signed` from ubuntu with `fwupd` from your repo?

mmstick, 2019-08-30:
Yes to both

phd21, 2019-09-07:
Dear System 76, Pop-os, etc...

Thank you for providing these firmware updating packages for other Linux distributions. There have been great reviews on your projects, unfortunately, there have been no good instructions on how to install this in other systems?

I tried the "git clone" methods which did not work for me. Then, I found this PPA. 

Can you please provide simple easy to understand instructions that other non-System76 Linux users could use to install and use this? I use Linux Mint 19.1 Cinnamon and KDE Neon user edition both are based on Ubuntu 18.04 Bionic. I already have the "fwupd" software installed, does that need to be removed first?

Example:
sudo apt-add-repository -ys ppa:system76-dev/stable
sudo apt-get update
sudo apt install fwupd firmware-manager 
or
sudo apt install fwupd fwupd-amd64-signed-template firmware-manager firmware-manager-notify firmware-manager-shared

Best regards,
Phil (phd21)

mmstick, 2019-09-07:
Installing `firmware-manager` and performing an `apt full upgrade` is enough.

phd21, 2019-09-08:
Hi mmstick,

Thank you for your fast reply. 

From all that I have read over the years, System76 produces excellent Linux computers and now some great software as well. I wish I had (could afford) a System76 Linux laptop. 

Have a great day and night.

Phil (phd21)