pop-os/firmware-manager #68: Incorrectly reports out of date firmware as "new".

pckilgore, 2019-09-09:
## What happened: 
On login, I am told I have new firmware available, but I do not:
```console
$ fwupdmgr get-updates
No upgrades for Precision M5520 Thunderbolt Controller, current is 26.01: 21.00=older
No upgrades for Precision 5520 System Firmware, current is 0.1.16.0: 0.1.16.0=same, 0.1.15.0=older, 0.1.14.3=older, 0.1.12.1=older, 0.1.11.0=older
________________________________________________

Devices that have been updated successfully:

 • Precision 5520 System Firmware (0.1.12.1 → 0.1.16.0)
```

##What I expected to happen:
Nothing.

## What might be the problem:
It looks like there is an `!=` check, rather than `<` check here, and since I've applied a higher version of firmware through Dell than is available in the firmware repo, it's incorrectly reported as out of date.  I didn't dig very far here but it looks like `fwupdmgt` got it correct, so maybe there is a way to piggy back of that and leave all the version parsing out of this library. 

https://github.com/pop-os/firmware-manager/blob/219bd059844f6524176ddc19670a92b35fc24ad0/notify/src/main.rs#L38-L53

pckilgore, 2019-09-09:
ref: https://bugs.launchpad.net/ubuntu/+source/fwupd/+bug/1842673

mmstick, 2019-09-09:
Using `<` is insufficient. Our daemon already performs a human sort of versions, the notify binary simply isn't using the information from the signal it receives.

pckilgore, 2019-09-09:
Even easier.  Thanks for the quick fix this has been annoying me for a few weeks now!

mmstick, 2019-09-09:
I didn't know about the issue until today. Feel free to file issues sooner :stuck_out_tongue:

pckilgore, 2019-09-09:
Of course that's on me.  It wasn't criticism; it was sincere appreciation.

commit 0b49149b (2019-09-09): fix(notify): fwupd firmware is upgradeable only if upgradeable is set

Use the `upgradeable` field in a fwupd firmware signal to know when
available firmware is actually an upgrade over installed firmware.

Closes #68
PR #69 (2019-09-18): fix(notify): fwupd firmware is upgradeable only if upgradeable is set
Use the `upgradeable` field in a fwupd firmware signal to know when
available firmware is actually an upgrade over installed firmware.

Closes #68
commit 017a0db8 (2019-09-18): fix(notify): fwupd firmware is upgradeable only if upgradeable is set

Use the `upgradeable` field in a fwupd firmware signal to know when
available firmware is actually an upgrade over installed firmware.

Closes #68
closed 2019-09-18 by mmstick