pop-os/firmware-manager #76: non-system updates not installing

pachamaltese, 2019-09-30:
this keeps appearing after installing the update

[image]

nicoflojo, 2020-01-08:
having this problem as well with a thinkpad x1 carbon.
$fwupdmgr update not working as well

[image]

aral, 2020-01-23:
Just ran into the same issue while trying to update the firmware for my Logitech mouse. It failed and now the update is not showing.

Is there a way to debug this from terminal?

ellisonleao, 2020-01-25:
same issue here with an x1 carbon 6th gen

regulator-g, 2020-03-21:
Hi guys

I have done a years worth of BIOS updates in Ubuntu with no issues

I recently switched to POP os and a week later firmware update available... I click update and the computer restarts, boots normally without updating the firmware!

I have tried

-to update via terminal (fwupdmgr)

-boot order is not locked

-tried secure boot off

-cant see any clues during reboot or in terminal

I'm on a Lenovo x1 carbon.

this guy had the same issue..

https://www.reddit.com/r/pop_os/comments/fiycsq/firmware_update_message_keeps_popping_up_but/