pop-os/firmware-manager #89: Unable to update device firmware - no update button

pflanzenandi, 2021-01-02:
[image]

leviport, 2021-01-04:
Are you talking about the Unifying Receiver firmware? If an update button is not shown, it most likely means the firmware is already up to date. Expanding the drop-down should show you more version info, so that may be useful.

pflanzenandi, 2021-01-04:
Hmm, I would assume that only updates are shown in this panel. But Ok

closed 2021-01-04 by pflanzenandi

leviport, 2021-01-04:
Having access to the firmware version and changelog info is useful for some, so we don't really want to hide that. Perhaps there is a better way to indicate that a device is up-to-date though.