pop-os/firmware-manager #90: GTK fails to initialize

jonathanjsimon, 2021-02-10:
While this builds for me, I cannot run it. I get the following exception at runtime if I run with elevated permissions. Anybody else?

```
$ pkexec firmware-manager
Unable to init server: Could not connect: Connection refused

Backtrace omitted. Run with RUST_BACKTRACE=1 to display it.
Run with RUST_BACKTRACE=full to include source snippets.

The application panicked (crashed).
  failed to init GTK: BoolError { message: "Failed to initialize GTK", filename: "/home/jsimon/.cargo/registry/src/github.com-1ecc6299db9ec823/gtk-0.9.2/src/rt.rs", function: "gtk::rt", line: 129 }
in src/libcore/result.rs, line 1188
thread: main
```

Obviously, if I run it without elevated permissions, it opens but doesn't work. I tried recompiling with Rust 1.49 instead of 1.41 to no avail.

bflanagin, 2021-02-10:
Curious it works with sudo when I run

```
sudo com.system76.FirmwareManager
```
Could be an issue with pkexec.

yochananmarqos, 2021-02-27:
@jonathanjsimon Both AUR packages are updated to include a Polkit policy to fix that.

jacobgkau, 2021-03-23:
@yochananmarqos Thank you for maintaining those AUR packages! I've confirmed the issue on an Arch system before the change and confirmed that it's currently fixed, so I'm closing this.

closed 2021-03-23 by jacobgkau