pop-os/firmware-manager #120: failed to map sideband memory: Operation not permitted

GerardoGR, 2022-04-12:
When running the `firmware-manager` from the command line. I always see the following error:

```shell
$ sudo firmware-manager 
[INFO ] firmware_manager src/system76.rs:11: scanning for system76 devices
[ERROR] firmware_manager src/system76.rs:24: failed to download system76 changelog: calling Download method failed: "failed to map sideband memory: Operation not permitted (os error 1)"
```
@ahoneybun (of system76's support team) and I were not able to find out the root cause for this. I wonder if this prevents me from getting the latest firmware versions.

Also to note: I've installed `firmware-manager` via `paru -S firmware-manager` which in my machine uses `rust=1.60.0-1`

GerardoGR, 2022-04-12:
```
$ lsblk -f
NAME                FSTYPE      FSVER    LABEL UUID                                   FSAVAIL FSUSE% MOUNTPOINTS
nvme0n1                                                                                              
├─nvme0n1p1         vfat        FAT32    EFI   07CB-2B30                               408.4M    20% /boot
└─nvme0n1p2         crypto_LUKS 2              390ce3ac-6fcd-4941-a8a8-548f6ec43e5e                  
  └─cryptlvm        LVM2_member LVM2 001       ge5ecI-eGdt-esSI-ck13-SMXD-abSM-q2R1XR                
    ├─vg--main-swap swap        1              5ade5259-19d6-4f54-9f71-3e4042b57b69                  [SWAP]
    ├─vg--main-root ext4        1.0            e77d4874-5697-4ccf-98d4-675ad1cd155e     69.7G    24% /
    └─vg--main-home ext4        1.0            a2e25f97-a4c1-43eb-bf64-7e8ae13daa44    603.5G    18% /home

$ sudo pacman -Qi firmware-manager 
Name            : firmware-manager
Version         : 0.1.2-5
Description     : GTK application for managing system and device firmware.
Architecture    : x86_64
URL             : https://github.com/pop-os/firmware-manager
Licenses        : GPL3
Groups          : None
Provides        : firmware-manager-virtual
Depends On      : dbus  gtk3  openssl  libgudev  libfirmware-manager  polkit
Optional Deps   : None
Required By     : None
Optional For    : system76-driver
Conflicts With  : None
Replaces        : None
Installed Size  : 9.27 MiB
Packager        : Unknown Packager
Build Date      : Thu 31 Mar 2022 12:42:24 AM CEST
Install Date    : Tue 12 Apr 2022 08:26:09 PM CEST
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None

```

ahoneybun, 2022-04-12:
I'm on the same version and has the same build date:

```
sda               8:0    0 111.8G  0 disk  
├─sda1            8:1    0   488M  0 part  /boot
├─sda2            8:2    0  94.3G  0 part  
│ └─crypt-root  253:3    0  94.3G  0 crypt /
└─sda3            8:3    0    17G  0 part  

pacman -Qi firmware-manager
Name            : firmware-manager
Version         : 0.1.2-5
Description     : GTK application for managing system and device firmware.
Architecture    : x86_64
URL             : https://github.com/pop-os/firmware-manager
Licenses        : GPL3
Groups          : None
Provides        : firmware-manager-virtual
Depends On      : dbus  gtk3  openssl  libgudev  libfirmware-manager  polkit
Optional Deps   : None
Required By     : None
Optional For    : system76-driver
Conflicts With  : None
Replaces        : None
Installed Size  : 9.27 MiB
Packager        : Unknown Packager
Build Date      : Thu 31 Mar 2022 01:10:12 PM MDT
Install Date    : Fri 01 Apr 2022 05:15:40 AM MDT
Install Reason  : Explicitly installed
Install Script  : Yes
Validated By    : None
```

GerardoGR, 2022-06-04:
I think a better place for this issue is at https://github.com/pop-os/system76-firmware because the error is actually coming from the `system76-firmware-daemon`. I'll close this issue in favor of a similar issue there: https://github.com/pop-os/system76-firmware/issues/82. But for me this is still a current issue.

closed 2022-06-04 by GerardoGR