pop-os/firmware-manager #123: Updates that are available but cannot be installed due to requirements are not displayed

Yethal, 2022-04-29:
For UEFI updates on laptops fwupdmgr requires that the laptop lid is open and laptop is on AC power and disallows firmware installation if those conditions are not satisfied (for example if the laptop is on battery or is docked in clamshell mode).
That information is not displayed in the Firmware UI so users will not find out about the updates untl they plug their laptops to AC or unplug their laptops from docking stations.

leviport, 2022-07-13:
Fwupdmgr also will not install certain updates if the battery is below 50%, but since there is no messaging displayed, the user won't really know why that is unless they run `fwupdmgr get-devices` in a terminal (which most users likely won't do). 

I think it'd be great if a message was displayed saying when and why fwupd updates cannot be installed. Something like, "Updates cannot be installed when on battery power".

pentamassiv, 2022-08-26:
I believe there recently was a commit and now you can see the devices for which there are updates but the requirements to install them are not fulfilled.
https://github.com/fwupd/fwupd/commit/519e80b1eeb89670031ddf415c02bf618c52ea25
Looks like the blocker for this feature is gone.