pop-os/firmware-manager #130: Random panic! after few clicks in firmware-manager

sigseg5, 2022-11-11:
Some time ago I had this nasty bug. If I open the firmware-manager and click on the items on me, the application crashes after a few seconds.See backtrace and video for details.

[firmware.webm](https://user-images.githubusercontent.com/36568961/201354882-e7664756-28ae-4aa7-af8f-f48bcf23d538.webm)

Hardware: `ThinkPad T480, UHD Graphics 620`
OS: `Pop!_OS 22.04 LTS`
GNOME: `42.4`
Windowing System: `x11`

<details>
<summary>Backtrace</summary>

```
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: BoolError { message: "Invalid return value: expected (), got gboolean", filename: "/build/firmware-manager-fz78Cx/firmware-manager-0.1.2~1665000869~22.04~3b827d1/vendor/glib/src/closure.rs", function: "glib::closure", line: 365 }', /build/firmware-manager-fz78Cx/firmware-manager-0.1.2~1665000869~22.04~3b827d1/vendor/glib/src/object.rs:2968:50
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
fatal runtime error: failed to initiate panic, error 5
fish: Job 1, 'gnome-control-center' terminated by signal SIGABRT (Abort)
```

</details>

<details>
<summary>Full backtrace</summary>

```
unknown device key: Protocol (v): Variant("org.nvmexpress")
unknown device key: CompositeId (v): Variant("a45df35ac0e948ee180fe216a5f703f32dda163f")
unknown device key: VersionRaw (v): Variant(240)
unknown device key: Protocol (v): Variant("org.uefi.capsule")
unknown device key: VersionRaw (v): Variant(65558)
unknown device key: VersionLowestRaw (v): Variant(65558)
unknown device key: Protocol (v): Variant("org.uefi.capsule")
unknown device key: VersionRaw (v): Variant(3093106915)
unknown device key: VersionLowestRaw (v): Variant(1)
unknown device key: Protocol (v): Variant("org.uefi.capsule")
unknown device key: VersionRaw (v): Variant(65582)
unknown device key: VersionLowestRaw (v): Variant(65536)
unknown device key: VersionRaw (v): Variant(20547760350566940)
unknown device key: Protocol (v): Variant("com.intel.thunderbolt")
unknown device key: Protocol (v): Variant("org.uefi.capsule")
unknown device key: VersionRaw (v): Variant(33619976)
unknown device key: VersionLowestRaw (v): Variant(1)
unknown device key: Protocol (v): Variant("org.uefi.capsule")
unknown device key: VersionRaw (v): Variant(17040444)
unknown device key: VersionLowestRaw (v): Variant(1)
unknown device key: Protocol (v): Variant("org.uefi.capsule")
unknown device key: VersionRaw (v): Variant(4784148)
unknown device key: VersionLowestRaw (v): Variant(4784148)
unknown device key: CompositeId (v): Variant("a45df35ac0e948ee180fe216a5f703f32dda163f")
unknown device key: Protocol (v): Variant("org.uefi.dbx")
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
unknown release key: ReleaseId (v)
unknown release key: Created (v)
unknown release key: Locations (v)
unknown release key: Urgency (v)
thread '<unnamed>' panicked at 'called `Result::unwrap()` on an `Err` value: BoolError { message: "Invalid return value: expected (), got gboolean", filename: "/build/firmware-manager-fz78Cx/firmware-manager-0.1.2~1665000869~22.04~3b827d1/vendor/glib/src/closure.rs", function: "glib::closure", line: 365 }', /build/firmware-manager-fz78Cx/firmware-manager-0.1.2~1665000869~22.04~3b827d1/vendor/glib/src/object.rs:2968:50
stack backtrace:
   0:     0x7f98c2aca87d - <unknown>
   1:     0x7f98c2af049c - <unknown>
   2:     0x7f98c2ac4501 - <unknown>
   3:     0x7f98c2acc075 - <unknown>
   4:     0x7f98c2acbd91 - <unknown>
   5:     0x7f98c2acc606 - <unknown>
   6:     0x7f98c2acc4f7 - <unknown>
   7:     0x7f98c2acad74 - <unknown>
   8:     0x7f98c2acc229 - <unknown>
   9:     0x7f98c26dac73 - <unknown>
  10:     0x7f98c26dadc3 - <unknown>
  11:     0x7f98c26f62a5 - <unknown>
  12:     0x7f98c271da27 - <unknown>
  13:     0x7f98c4a04745 - g_cclosure_marshal_VOID__OBJECTv
  14:     0x7f98c4a23640 - g_signal_emit_valist
  15:     0x7f98c4a237a3 - g_signal_emit
  16:     0x7f98c4046a89 - <unknown>
  17:     0x7f98c41f0be5 - <unknown>
  18:     0x7f98c4a23640 - g_signal_emit_valist
  19:     0x7f98c4a237a3 - g_signal_emit
  20:     0x7f98c4007ffc - <unknown>
  21:     0x7f98c4a0a866 - g_cclosure_marshal_VOID__BOXEDv
  22:     0x7f98c4a23640 - g_signal_emit_valist
  23:     0x7f98c4a237a3 - g_signal_emit
  24:     0x7f98c3fffacb - <unknown>
  25:     0x7f98c400783b - <unknown>
  26:     0x7f98c4008443 - <unknown>
  27:     0x7f98c3fcef90 - gtk_event_controller_handle_event
  28:     0x7f98c41a0045 - <unknown>
  29:     0x7f98c41e6eb8 - <unknown>
  30:     0x7f98c4a23640 - g_signal_emit_valist
  31:     0x7f98c4a237a3 - g_signal_emit
  32:     0x7f98c41ae724 - <unknown>
  33:     0x7f98c4051680 - <unknown>
  34:     0x7f98c405252a - gtk_main_do_event
  35:     0x7f98c473d743 - <unknown>
  36:     0x7f98c4774f56 - <unknown>
  37:     0x7f98c490cd1b - g_main_context_dispatch
  38:     0x7f98c49616f8 - <unknown>
  39:     0x7f98c490a3c3 - g_main_context_iteration
  40:     0x7f98c4b31c85 - g_application_run
  41:     0x55d9d7530d83 - main
  42:     0x7f98c1629d90 - __libc_start_call_main
                               at ./csu/../sysdeps/nptl/libc_start_call_main.h:58:16
  43:     0x7f98c1629e40 - __libc_start_main_impl
                               at ./csu/../csu/libc-start.c:392:3
  44:     0x55d9d7530f65 - _start
  45:                0x0 - <unknown>
fatal runtime error: failed to initiate panic, error 5
fish: Job 1, 'RUST_BACKTRACE=full gnome-contr…' terminated by signal SIGABRT (Abort)
```

</details>

mjbain10, 2022-12-14:
I am experiencing this as well. Backtrace is identical. 

Hardware: ThinkPad X1 Carbon Gen9
OS: Pop!_OS 22.04 LTS
GNOME: 42.5
Windowing System: Wayland
Kernel: Linux 6.0.6-76060006-generic

mjbain10, 2022-12-14:
I've noticed that when clicking slowly, this issue does not happen. It only happens when clicking/expanding different firmware sections in rapid succession.

reuvenpo, 2023-12-26:
I'm seeing a similar issue to this today. looking through systemd logs (using gnome-logs) i see the following messages:

```rust 
called `Result::unwrap()` on an `Err` value: 
BoolError {
    message: "Invalid return value: expected (), got gboolean",
    filename: "/build/firmware-manager-wb87IC/firmware-manager-0.1.2~1701473591~22.04~fc93648/vendor/glib/src/closure.rs",
    function: "glib::closure",
    line: 365
}
```
(formatted for readability)

and
```
thread '<unnamed>' panicked at /build/firmware-manager-wb87IC/firmware-manager-0.1.2~1701473591~22.04~fc93648/vendor/glib/src/object.rs:2968:50:
```

PR #152 (2024-01-20): fix: set the correct return type in `emit_by_name` to prevent `panic!` after few fast clicks on firmware-manager items
Closes #130
commit 110d9954 (2024-01-20): fix: set the correct return type in `emit_by_name` to prevent `panic!`

Closes #130
closed 2024-01-20 by mmstick